Chaostreff-Potsdam/erika3004 #23: Tracking Erikas Cursor Position

sirexeclp, 2019-08-17:
How can we track the current cursor position of erika, if we need absolute values?
e.g. every method that moves / writes has to update an internal cursor position value
Or can we get the cursor position from erika?
Or can we use some kind of enstop to "home" before making absolute cursor moves.
For columns this could be done using carriage return, for lines this might be difficult.

Any ideas ?

carwe, 2019-08-24:
Also this is needed to be able to prohibit moving above a specific line when printing data from untrusted sources to avoid destroying text which wasn't printed by the same person.

cyroxx, 2020-01-07:
From the serial port, the device has no means to track the cursor position. So unless you want to modify the internal electronics, you have to do the tracking/counting in software.

carwe, 2020-01-07:
iirc this was discussed as feature of some layer between network and serial port - not sure anymore if this project is the correct one for this feature request

sirexeclp, 2020-01-07:
This is needed to make some games work.
Current status is to implement it in software, as i have done for the TicTacToe game.
But it would be better suited to be in the Erika main class.

sirexeclp, 2020-01-07:
@carwe btw. if you want one or more erikas we now a few "in stock" so to say ;)
see https://github.com/Chaostreff-Potsdam/erika3004/wiki/Compatible-Typewriters