Chaostreff-Potsdam/erika3004 #32: TCP Server: allow full 65 characters

cyroxx, 2020-01-03:
We currently enforce a line-wrap after 65 characters, whatever the characters are. So if somebody sends us lines of 65 characters + newline, we actually get 2 newlines: the one that is enforced and the one from the incoming request.

We should skip enforcing line wrap in that case since the client already sends us the newline.

ArchibaldBienetre, 2020-01-18:
There's a similar Issue request by @sirexeclp 
https://github.com/Chaostreff-Potsdam/erika3004/issues/73
+ follow-up https://github.com/Chaostreff-Potsdam/erika3004/issues/74

cyroxx, 2020-01-18:
btw, the default number of characters is 78. The 65 characters was just a setting that i used at some point for demonstration purposes when the device is unattended.
The enforced line-wrap (see above) still needs to be eliminated, though.

sirexeclp, 2020-01-19:
see http://hc-ddr.hucki.net/wiki/doku.php/z9001/erweiterungen/s3004

Code | Funktion | Bemerkung
-- | -- | --
80H | Randlöser | Rand lösen
8FH | Randlöser ON (extern) | Rand rechts öffnen (*); Extremer Randlöser. Die Randeinstellung wird bis zum Eintreffen von SETRD nicht mehr berücksichtigt.

sirexeclp, 2020-01-19:
also i've copy pasted the table into our wiki, so we can add our own comments, to make things clearer
and ofc. translate it
https://github.com/Chaostreff-Potsdam/erika3004/wiki/Control-characters