Chaostreff-Potsdam/erika3004 #34: Erika + Erika mocks: support "delete"

ArchibaldBienetre, 2020-01-05:
We need to support the correction tape, e.g. for #29 (Tetris).

It seems one control character is there to switch from correction to normal tape, one to switch back. 
The actual deletion requires the implementer to keep track of what letter was there before - and overwrite it with with itself using the correction tape.

See #29: 

> Correction tape controls are not yet wrapped by our software - the website linked to by the main README.md provides some insights here, though:  
> ```
> 8BH	Zeichen löschen OFF	Status: Drucken (die nachfolgenden Zeichen werden gedruckt)
> 8CH	Zeichen löschen ON	Status: Korrigieren (die nachfolgenden Zeichen werden mit Korrekturband gelöscht)
> ```
> http://hc-ddr.hucki.net/wiki/doku.php/z9001/erweiterungen/s3004

sirexeclp, 2020-01-05:
this works

>>> import erika.erika
>>> e=erika.erika.Erika("/dev/ttyAMA0")
>>> e.move_right()
>>> e.move_right()
>>> e.move_left()
>>> e.move_left()
>>> e.print_ascii("M")
>>> e.move_left()
>>> e.move_left()
>>> e.print_ascii("M")
>>> e.print_ascii(" ")
>>> e.print_ascii(" ")
>>> e.print_ascii(" ")
>>> e.print_ascii(" ")
>>> e.print_ascii("M")
>>> e.move_left()
>>> e.print_ascii("M")
>>> e.print_ascii("Hallo")
>>> e._print_raw("8C")
>>> e.move_left()
>>> e.print_ascii("o")
>>> e.move_left()
>>> e.move_left()
>>> e.print_ascii("l")
>>> e.move_left()
>>> e.move_left()
>>> e.print_ascii("l")
>>> e.move_left()
>>> e.move_left()
>>> e.print_ascii("a")
>>> e.move_left()
>>> e.move_left()
>>> e.print_ascii("H")

ArchibaldBienetre, 2020-01-14:
I'd be glad if we could merge this issue this Wednesday.

https://github.com/Chaostreff-Potsdam/erika3004/pull/63 depends on this branch and is good for merge already as well.

ArchibaldBienetre, 2020-01-15:
Fixed! 
Other rework goes to https://github.com/Chaostreff-Potsdam/erika3004/issues/60

closed 2020-01-15 by ArchibaldBienetre