Chaostreff-Potsdam/erika3004 #36: Test ESP8266/32

sirexeclp, 2020-01-06:
Does the ESP support RTS/CTS, can we use it to control the erika?

ArchibaldBienetre, 2020-01-18:
We could not use ESP in the end because of a limited buffer

Can we rule out such issues here (or am I totally off track? :D )

> ## FT232RL
> The USB-to-serial adapter FT232RL supports hardware control flow (using the RTS/CTS lines), however, it can take up to three characters after the Erika sends its busy signal until the adapter stops sending data. 
> 
> > If CTS# is logic 1 it is indicating the external device cannot accept more data. the FTxxx will stop transmitting within 0~3 characters, depending on what is in the buffer.
> >
> > This potential 3 character overrun does occasionally present problems.
> 
> source: https://www.ftdichip.com/Support/FAQs.htm
> 
> For our case, this is not enough. 
https://github.com/Chaostreff-Potsdam/erika3004/wiki/Hardware-control-flow-(RTS,-CTS)#ft232rl

sirexeclp, 2020-01-18:
This is mostly an hardware TODO issue.

cyroxx, 2020-01-18:
@sirexeclp From the different specs that you quoted, ESP32 seems to be the better choice than its predecessor ESP8266. Because of its limitations (your second quote and link), it would be cumbersome to get it to work on a ESP8266 (only UART0 has RTS/CTS and we would need to find a way to suppress information that is printed during boot).