Chaostreff-Potsdam/erika3004 #47: Test ESP w/ 5V bus on Erika

sirexeclp, 2020-01-06:


cyroxx, 2020-01-07:
Please note that the ESP has an operating voltage range of 2.5 V ~ 3.6 V.

Kaibu, 2020-01-11:
5V bus seems to supply too little current for a raspberry pi zero, a esp would probably work neither.
The 17V bus with a step down module to 5/3V works fine though

ArchibaldBienetre, 2020-01-18:
Does this overlap with #36 ?

sirexeclp, 2020-01-18:
> 5V bus seems to supply too little current for a raspberry pi zero, a esp would probably work neither.
> The 17V bus with a step down module to 5/3V works fine though

Indeed the supply voltage dropped below 5V IIRC.
It might work though with using a buck-converter instead of the builtin low dropout linear regulator.

ArchibaldBienetre, 2020-01-18:
So, close, and link to #40 for continuation? :)

sirexeclp, 2020-01-18:
#40 is already closes; this is different as it would not require to modifications inside erika