Chaostreff-Potsdam/erika3004 #66: Print unknown characters using microsteps

ArchibaldBienetre, 2020-01-15:
We will likely never get a type wheel (Typenrad, = disc with Erika's character set) that has relatively new symbols, such as 
* the Euro sign
* emoji

As a workaround, we could instead use Erika's microstep feature, which is already wired up, tested and working.

sirexeclp, 2020-01-17:
€ could be printed as C and = overlay do we have a `=` ?

ArchibaldBienetre, 2020-01-18:
Good point, we could use other hacks to print unknown characters.
That should be a separate issue, though, imo.

We could apply a multi-step fallback - try the character combination hack first (for EUR, Yen, ... Pound, simple Emoji?), then resort pixelwise printing as suggested here.