Chaostreff-Potsdam/erika3004 #70: Erika CLI: move specific imports into funcctions

ArchibaldBienetre, 2020-01-15:
On a RasPi, it's not as easy to just install all requirements from requirements.txt as on a "normal" machine.

E.g., the CLI will fail if the Python Imaging Library (pillow) is missing - even though the user may not even need it.

To solve this, move import statements to the specific methods of cli.py

ArchibaldBienetre, 2020-01-16:
I tried around a bit - if a module uses an import (like `import numpy as np`), this would have to be moved into every method that uses it. 

This sounds a bit weird... and we may miss a lot of imports here.

Instead, I'll rather work closely with the RasPi crowd and identify which imports to remove, specifically.
@cyroxx and @sirexeclp can already give me some pointers, e.g. PIL / Pillow was mentioned (see initial comment)

sirexeclp, 2020-01-16:
moving imports to methods makes only sense  for the cli module, where multiple subprograms come together that can work independent of each other
everything that is always needed can be a global import
in the "sub-modules" imports can and in fact should be global

ArchibaldBienetre, 2020-01-16:
Got it - I overlooked that the `erika.cli` module also tries loading a `erika.image_converter.WrappedImage` to guess if an input image is binary or ASCII art.
See `erika.cli.get_erika_for_given_args`

Confusing: **There is no import for PIL nor `erika.image_converter.WrappedImage` in `erika.cli`.** (though this absolutely works.  
I guess I need to challenge my knowledge and assumptions regarding python imports :thinking:

sirexeclp, 2020-01-16:
right i guess `from X import *` loads the whole module, which executes all statements i.e. other import statements in this module
`from X import Y` only loads the specified function or class
btw thats why people write `if __name__ == "__main__"`
because without it the code would be executed, when the module is imported

closed 2020-01-22 by sirexeclp