Chaostreff-Potsdam/erika3004 #75: Erika as a Service

sirexeclp, 2020-01-18:
How about setting up one Erika with computer-paper and a webcam, that is accessible from the internet (or vpn) so that one could test things on real hardware if he/she does not have an Erika at hand.
Ofc. access should be limited in some way and some safety features might be implemented such as remotely restarting Erika.

ArchibaldBienetre, 2020-01-18:
Awesome label added :) 

That requires a whole new threat model :D 

How confident are we about the TCP server - should we test some possibly malicious cases?

Should we expose a binary interface at all, or just offer the wrapped public methods of the python interface? :thinking:

Kaibu, 2020-01-18:
I could take on that task and set that up at the hackspace.
I would make it only accessible via a vpn via server that the erika pi would autologin on and build in a relais to power cycle the erika from the pi.
I wouldn't make it public from the internet altough that would be an option later.

sirexeclp, 2020-01-18:
I guess a pi that's accessible via ssh behind a vpn would be fine as first
step.